Seeed-Studio/ModelAssistant #20: Firmware for Grove AI is 0 Byte

gbcarlos, 2023-01-16:
Hi,

I realized the firmware for the Grove Vision AI Sensor has a size of 0 Byte. When copying the fw and the model the sensor disconnects correctly after saving the files. However, I suspect that the firmware has not been updated and is still the old one:

- If I upload a firmware with which I am not able to see the camera output with and then upload your firmware and model afterwards I also can't see the camera output
- However if I upload a firmware that allows me to see camera output on the webtool and then upload your firmware and model I'm able to see the output

Also, if I see output I only see the image without bounding boxes or any detection result.
Might this be due to the firmware or am I overlooking something?

LynnL4, 2023-01-17:
Hi, Can you tell us which model and firmware you are using? And, what you mean `firmware for the Grove Vision AI Sensor has a size of 0 Byte`

LynnL4, 2023-01-18:
Oh, The firmware is working on [grove visino ai](https://www.seeedstudio.com/Grove-Vision-AI-Module-p-5457.html). Wio terminal does not have the Vision function.  And we will update the firmware link later. Thanks!

LynnL4, 2023-01-18:
And， this firmware only needs Grove Vision Ai to work independently.

gbcarlos, 2023-01-18:
I'm sorry I'm aware of that and I have been using the sensor independently so far, mentioning the model of the Wio Terminal was a mistake. However, I have not been able to find the model of the Grove Vision AI sensor, the issue with the firmware remains the same. Looking forward to the updated version.

LynnL4, 2023-01-29:
Hi, meter [model](https://files.seeedstudio.com/wiki/Edgelab/uf2/analog-meter-model.uf2)

closed 2023-01-29 by LynnL4