Seeed-Studio/ModelAssistant #117: Feature Request: Docs to show how to use the Arduino IDE to install the EdgeLab bootloader and also how to get back to Arduino

hpssjellis, 2023-08-10:
**Feature Request: Docs to show how to use the Arduino IDE to install the EdgeLab bootloader and also how to get back to the Arduino XIAO bootloader**

**Motivation**
Many of the people I teach will want to try this method of using the XIAO-Esp32-S3-Sense for machine learning but will also want to go back to the original shipped version of the Arduino Bootloader on the XIAO.

**Issue**
Without the EdgeLab bootloader I can not connect my XIAO-Esp32-S3-Sense to either:

The web-app version    
https://seeed-studio.github.io/edgelab-web-app/#/dashboard/workplace

or The ESP flash tool   
https://espressif.github.io/esptool-js/

**Related resources**
My version of using a web browser to serial connect to an Arduino like MCU and do machine learning is at  https://hpssjellis.github.io/tinyMLjs/public/index.html

**Additional context**
I have met the Seeedstudio founder Eric Pan, he will know me as Jeremy Ellis from the [ICTP conference in Trieste Italy](https://tinyml.seas.harvard.edu/EdgeMLUP-23/).

Thank you for considering my request.

LynnL4, 2023-08-11:
Hi, Thanks for your suggestion, we are working on a libraries that will enable EdgeLab trained models to be quickly deployed to devices and can be configured as well as visualized using a web-app.
https://github.com/Seeed-Studio/edgelab-example-esp32/pull/4

LynnL4, 2023-08-11:
Hi,
Regarding the issue of requiring the Edgelab bootloader to connect to the tool, it seems that it is not necessary for https://espressif.github.io/esptool-js/. This tool is used for flashing the corresponding bootloader and application.
However, for https://seeed-studio.github.io/edgelab-web-app/#/dashboard/workplace, connecting to the tool requires flashing a specific firmware. This firmware will be released in the near future.

hpssjellis, 2023-08-11:
Thank you @LynnL4 for your fast reply. On my Ubuntu Linux computer I can't seem to connect using the esp-flash tool. I will try next week on a windows computer. 

Yes, if that connects then installing the correct bootloaders will be fine. Will you also soon publish the bootloader needed to get the XIAO working again with the Arduino IDE or do we get that from the espressif site?

LynnL4, 2023-08-11:
In my testing, macOS, Ubuntu, and Windows should all work fine. However, there might be some access permission issues on Ubuntu. If you could provide the log detailing the issues you encountered, we can analyze them further. Additionally, based on my experience, returning to Arduino for development shouldn't require any special actions. The Arduino IDE will reflash the bootloader automatically. If it doesn't work, you can try holding the BOOT button (or reset) while powering on the device before flashing.

hpssjellis, 2023-08-11:
@LynnL4 my mistake. I was using Chromium on Ubuntu which does not do webSerial. When I switched to MS Edge both web-apps connected fine. On the tutorial site https://wiki.seeedstudio.com/tinyml_workshop_course/#21-quick-hands-on-experience it has the binaries to flash to the XIAO-sense so I am good there. Now my only concern is how to get the board back to the Arduino bootloader. Those binaries must be available somewhere. I will try to find them.

hpssjellis, 2023-08-11:
@LynnL4 you are correct about getting back to Arduino. Yes I could easily flash a regular Arduino program after using the ESP-tool.  I did manage to flash the tutorial examples the digital reader, face detection and sound detection but only once seemed to get anything working on EdgeLab. The face detection program camera was working after I picked the model, but it never detected my face. I think that proves that the esp-tool was working so for now I can close this issue. I never did get the sound or digital reader working as EdgeLab did not seem to find a model. I will wait a bit and log another issue when I am at a computer that makes it easy to upload images.

By the way, my repository of XIAO-Esp32-S3  working examples is here  https://github.com/hpssjellis/mcu-stable-edu-launch/tree/main/xiao-esp32s3  I even got a lidar detector working. 

Thanks so much @LynnL4 for your help.

closed 2023-08-11 by hpssjellis

hpssjellis, 2023-08-12:
@LynnL4 I did get everything working using ubuntu linux and the Edge Browser. One tip I used was in the ESP-flash tool I connected the board after doing a simple reset (not DFU mode)  and then chose "erase flash" first. When that was successful  I loaded the files from the tutorial.

Too bad the esp-tool does not except JSON files to auto load each of the 4 files. That might be an easy edit to the esp-tool github.

Images of working projects.
Digits
[image]

Face
[image]

The Hello Voice project worked for me but only in the serial monitor not using the edgeLab.

A very exciting project you are working on. Well done.

LynnL4, 2023-08-12:
Thanks for sharing your usage, we are in the process of integrating esptool into edgelab-web-app, which will simplify firmware burning. In addition, we will continue to update the support for models, such as image classification, sound recognition, sensor classification and so on. Please also contact us if you have any questions or suggestions for subsequent use.

hpssjellis, 2023-08-12:
@LynnL4 Integrating the esptool is interesting. You probably could make a text or json textarea or file upload that could set the address and file names so all 4 files could be loaded into the tool in one command.

Something else to think about is the ability to convert TensorflowJS files to TFLITE files. I have asked edgeImpulse to allow it but they seem slow to realise why it is useful.   Super easy with the backend you will have. Really hard for me to do using just front-end javascript.

https://forum.edgeimpulse.com/t/feature-request-be-able-to-upload-a-tensorflowjs-model/8027

hpssjellis, 2023-08-14:
@LynnL4  I just checked the esp-tool and it will not load a binary from a URL. Probably for obvious security issues, but for a conference a text file with the internet locations of all the needed data would be very useful and should be easy to code in Javascript. If the binary can be loaded from a url then all 4 downloads could be uploaded to the esp-tool in one command. A textarea with text input such as the following would do the trick: This file could also be daved on the users computer for later reference.

```
"flashAddress":"0x0000",
"fileName":"bootloader.bin",
"locationOrUrl":"https://files.seeedstudio.com/wiki/tinyml-topic/res/bootloader.bin";

"flashAddress":"0x8000",
"fileName":"partition-table.bin",
"locationOrUrl":"https://files.seeedstudio.com/wiki/tinyml-topic/res/partition-table.bin";

"flashAddress":"0x10000",
"fileName":"edgelab.bin",
"locationOrUrl":"https://files.seeedstudio.com/wiki/tinyml-topic/res/edgelab.bin;

"flashAddress":"0x400000",
"fileName":"face_detection.tflite",
"locationOrUrl":"https://files.seeedstudio.com/wiki/tinyml-topic/face_detection.tflite";

```

I hope this is a better explanation than my first atempt.

LynnL4, 2023-08-15:
I agree with you, now I am checking if I can load the firmware from the URL.  Thanks for the suggestion.