Seeed-Studio/ModelAssistant #138: Steps to make the binaries using the Arduino IDE and install them using SenseCraft

hpssjellis, 2023-09-04:
**Steps to make the binaries using the Arduino IDE and install them using SenseCraft**

@LynnL4 

I always start projects from the basics. I would like to take a simple LED and Serial Print program and get in running on SenseCraft. Exporting the binaries for the app (0x10000), partition (0x8000) and bootloader (0x1000) from the Arduino IDE is easy, but installing the binaries using [SenseCraft](https://seeed-studio.github.io/SenseCraft-Web-Toolkit/#/dashboard/workplace) or the [ESP tool](https://espressif.github.io/esptool-js/) does not seem to work.  

If with some suggestions I can get that simple installation working I would then like to look into the edgeLab app and see how to get serial output working with SenseCraft.

P.S. @LynnL4 do you have my Arduino Library installed?  The Portenta Pro Community Solutions? I will probably make a XIAO version of this library but most of the code already works with a few pin changes. The basic starting program I always use is called    [dot11-hello-blink.ino](https://github.com/hpssjellis/portenta-pro-community-solutions/blob/main/examples/dot1-portentaH7-examples/dot11-hello-blink/dot11-hello-blink.ino)   I would like to get it working first with just serial output and then with edgeLab so the serial output shows up on the senseCraft screen.

hpssjellis, 2023-09-04:
@LynnL4 

I got it working using the esp flash tool, loading the bootloader at 0x000 the partition table at 0x8000 and the app at 0x10000 and that worked fairly smooth. It says an error with the bootloader but still works. If I don't load the bootloader it fails. location 0x1000 for the bootloader fails.

Still having no luck with SenseCraft. I will however start looking at the edgelab app

hpssjellis, 2023-09-05:
@LynnL4 can you or "Serena" send me the edgelab.ino. I can't seem to find it on the github.

LynnL4, 2023-09-05:
Hi, We are building more stable firmware, please wait(probably available in the next week). I'll send it to you when it's finished developing, thanks!

LynnL4, 2023-09-14:
Hi, We have updated the SSCMA Web Toolkit, enabling faster deployment of the model on the Xiao ESP32-S3.

https://wiki.seeedstudio.com/edgelab/
https://seeed-studio.github.io/SenseCraft-Web-Toolkit/#/setup/process

hpssjellis, 2023-09-14:
@LynnL4 That is amazing! Well done, so easy and fast.

closed 2024-03-05 by hpssjellis