Seeed-Studio/ModelAssistant #180: Checking that the ONXX export is successful for pfld_mbv2n_112.py fails with AttributeError

sbocconi, 2024-03-13:
**Describe the bug**
Checking that the ONXX export is successful for `configs/pfld/pfld_mbv2n_112.py` fails with `AttributeError: 'MeterData' object has no attribute '_metainfo'. Did you mean: 'metainfo'?`

**Environment**
Environment you use when bug appears:

1. Python version: 3.10
2. PyTorch Version: torch==2.0.1
3. MMCV Version: 2.0.1
4. EdgeLab Version: na
5. Code you run
`python3 tools/inference.py configs/pfld/pfld_mbv2n_112.py work_dirs/pfld_mbv2n_112/epoch_1_float32.onnx --cfg-options data_root=datasets/meter/`

6. The detailed error
```
Traceback (most recent call last):
  File "/Users/SB/Projects/Software/Zephyros/Courses/Microcontrollers/ModelAssistant/tools/inference.py", line 352, in <module>
    main()
  File "/Users/SB/Projects/Software/Zephyros/Courses/Microcontrollers/ModelAssistant/tools/inference.py", line 336, in main
    runner = Infernce(
  File "/Users/SB/Projects/Software/Zephyros/Courses/Microcontrollers/ModelAssistant/sscma/utils/inference.py", line 289, in __init__
    self.class_name = dataloader.dataset._metainfo['classes']
AttributeError: 'MeterData' object has no attribute '_metainfo'. Did you mean: 'metainfo'?
```
**Additional context**
Running on Mac M2, torch cpu-only, mmcv compiled from source

MILK-BIOS, 2024-03-14:
Hi @sbocconi, this problem is mainly caused by using the model `pfld` and the point meter dataset which **NOT** contains any category. We fix it in the latest code. 
If you have other questions, please contact us. Thank you for your support!

mjq2020, 2024-03-19:
Hi @sbocconi,We have fixed this bug. You can pull the latest version of the code to test it.

MILK-BIOS, 2024-03-19:
> Hi @MILK-BIOS I am not sure I understand, is the bug fixed somewhere and can I get that version? Thanks

Sorry for the delay in releasing our code. You can check it now!

sbocconi, 2024-03-19:
This works, thanks!

closed 2024-03-19 by sbocconi