Seeed-Studio/ModelAssistant #250: setup at colab and face error

kris-himax, 2024-08-06:
**Describe the bug**
While setup the ModelAssistant at colab, facing following error.
[image]
I think that it is the error at installing ethos-u-vela 3.3.0. But there have been installing latest vela version, why should it be installed older version about vela?

**Environment**
Environment you use when bug appears:
using the default colab script https://colab.research.google.com/github/seeed-studio/sscma-model-zoo/blob/main/notebooks/en/person_Detection_Swift-YOLO_192.ipynb
at  https://wiki.seeedstudio.com/ModelAssistant_Introduce_Quick_Start/#getting-started 

Thanks,
Kris

iChizer0, 2024-08-07:
Thank you for your feedback, this may be caused by pip handling multiple dependencies at the same time as well as installing these dependencies again when running the setup.py, I've created a patch after testing it and I hope it solves your problem.

1. Save the following to `fix.patch` to the `ModelAssistant` home directory

```
diff --git a/scripts/setup_colab.sh b/scripts/setup_colab.sh
index 97f70d5..ed3cc4b 100755
--- a/scripts/setup_colab.sh
+++ b/scripts/setup_colab.sh
@@ -26,7 +26,10 @@ pip install torch==2.0.0 torchvision==0.15.1 torchaudio==2.0.1
 
 # install base deps
 echo -en "Installing base deps... "
-pip install -r requirements/base.txt -r requirements/inference.txt -r requirements/export.txt -r requirements/tests.txt
+pip install -r requirements/base.txt
+pip install -r requirements/inference.txt
+pip install -r requirements/export.txt
+pip install -r requirements/tests.txt
 
 if [ "$?" != 0 ]; then
     echo -en "Install base deps failed... ${RED}Exiting${RST}\n"
@@ -38,11 +41,14 @@ fi
 echo -en "Installing OpenMIM deps... \n"
 mim install -r requirements/mmlab.txt && \
 mim install mmcv==2.0.0 && \
-mim install -e .
+
 if [ "$?" != 0 ]; then
     echo -en "OpenMIM install deps failed... ${RED}Exiting${RST}\n"
     exit 1
 fi
 
+echo -en "Installing SSCMA\n"
+pip install -e .
+
 echo -en "Finished setup... ${GREEN}OK${RST}\n"
 exit 0
diff --git a/setup.py b/setup.py
index 840ef7c..a8b8c7a 100644
--- a/setup.py
+++ b/setup.py
@@ -128,7 +128,7 @@ if __name__ == '__main__':
             'Topic :: Software Development',
             'Topic :: Scientific/Engineering',
         ],
-        install_requires=requirements,
+        # install_requires=requirements,
         include_package_data=True,
         python_requires='>=3.8',
         license='Apache License 2.0',
```

3. Apply the patch in Colab

```
!git apply fix.patch
```

kris-himax, 2024-08-07:
Hi @iChizer0 ,

I add the patch by
```
!git apply fix.patch
```
but face the error.
[image]

iChizer0, 2024-08-07:
@kris-himax It looks like it's due to a missing new line at the end of the copied patch, you can either manually  or run the following command `!sed -i -e '$a\' fix.patch` to add a blank line  before applying it.

kris-himax, 2024-08-07:
Hi @iChizer0 ,
It works fine.

Thanks for reply.

iChizer0, 2024-08-07:
@kris-himax We need to do some testing to make sure it doesn't affect other examples, so updating to the main branch may take some time, we'll take care of it as soon as possible, thanks.

closed 2024-08-07 by kris-himax