Seeed-Studio/ModelAssistant #253: !!URGENT!!:  Code not working

abhijeetGS, 2024-08-19:
**Describe the bug**
In "Export the model" section, there is a reference to /last_checkpoint file which is not mentioned anywhere prior in the code. This is raising a "FileNotFound" error

import os
with open('Apple_Detection_Swift-YOLO_192/last_checkpoint', 'r') as f:
	os.environ['CHECKPOINT_FILE_PATH'] = f.read()

**Additional context**
This is a part of the elaborate code inorder to build a custom model for "Grove vision AI" module. The link to the page is [here](https://wiki.seeedstudio.com/grove_vision_ai_v2_sscma/) and the link to the colab code(s) is [here](https://wiki.seeedstudio.com/ModelAssistant_Introduce_Quick_Start/#model-training)

I tried running the Apple_Detection_Swift-YOLO_192 as is and got the same above error. Please help

iChizer0, 2024-08-22:
Hello, thanks for your feedback.

I've tested the same notebook but I cannot reproduce your issue, could you please provide your training log or current path in the Colab? Thanks.

baorepo, 2024-09-13:
I see that you have not responded for a long time, I will close this issue first, if you have new developments, we will open this issue.

closed 2024-09-13 by baorepo