Seeed-Studio/ModelAssistant #276: Custom classification model: Exported tflite model fails to invoke in Grove Vision AI V2 boards

rn776, 2025-01-15:
I am training a binary classification model using the SSCMA library using a custom dataset with images of two classes. I have used the fomo_mobnetv2_0.1_x8_coco.py from the examples as the template. I have managed to export the trained .pth model to tflite format. However, after uploading it using the SenseCraft AI web toolkit, it shows a 'invoke failed' output. Please let me know if there an error in the training configuration or any other area that needs to be investigated. My training config file is:
```
from mmengine.config import read_base
from mmengine.dataset import DefaultSampler

with read_base():
    from .._base_.default_runtime import *
    from .._base_.schedules.schedule_1x import *

from sscma.datasets import CustomDataset
from mmengine.optim import LinearLR, MultiStepLR

from sscma.datasets.transforms import (
    RandomResizedCrop,
    PackInputs,
    LoadImageFromFile,
    RandomFlip,
    ResizeEdge,
    CenterCrop

)
from sscma.engine import DetVisualizationHook
from sscma.visualization import UniversalVisualizer
from sscma.datasets import ClsDataPreprocessor

from sscma.models import ImageClassifier, CrossEntropyLoss, GlobalAveragePooling
from sscma.models import MobileNetv2, LinearClsHead

from sscma.evaluation import Accuracy

num_classes = 2
widen_factor = 0.1
dataset_type = CustomDataset
data_root = "/home/pipe_dataset"

metainfo = None

height = 192
width = 192
imgsz = (width, height)

downsample_factor = (8,)
batch = 8
workers = 2
persistent_workers = True

val_batch = 16
val_workers = 2

lr = 0.02
epochs = 50

weight_decay = 0.0005
momentum = 0.95

default_hooks = dict(visualization=dict(type=DetVisualizationHook, score_thr=0.8))
visualizer = dict(type=UniversalVisualizer)
imdecode_backend = "cv2"

train_pipeline = [
    dict(type=LoadImageFromFile,imdecode_backend='cv2'),
    dict(type=RandomResizedCrop, scale=192),
    dict(type=RandomFlip, prob=0.5, direction="horizontal"),
    dict(type=PackInputs),
]

test_pipeline = [
    dict(type=LoadImageFromFile,imdecode_backend='cv2'),
    dict(type=ResizeEdge, scale=256, edge="short"),
    dict(type=CenterCrop, crop_size=192),
    dict(type=PackInputs),
]

train_dataloader = dict(
    batch_size=batch,
    num_workers=workers,
    persistent_workers=persistent_workers,
    drop_last=False,
    # pin_memory=True,
    dataset=dict(
        type=dataset_type,
        data_root=data_root,
        data_prefix=dict(img_path="train"),
        pipeline=train_pipeline,
    ),
    sampler=dict(type=DefaultSampler, shuffle=True),
)

val_dataloader = dict(
    batch_size=val_batch,
    num_workers=val_workers,
    persistent_workers=persistent_workers,
    drop_last=False,
    # pin_memory=True,
    dataset=dict(
        type=dataset_type,
        data_root=data_root,
        data_prefix=dict(img_path="valid"),
        pipeline=test_pipeline,
    ),
    sampler=dict(type=DefaultSampler, shuffle=False),
)

val_evaluator = dict(type=Accuracy)

test_dataloader = val_dataloader
test_evaluator = val_evaluator

find_unused_parameters = True

data_preprocessor = dict(
    type=ClsDataPreprocessor,
    mean=[0, 0, 0],
    std=[255.0, 255.0, 255.0],
    to_rgb=True,
    num_classes=num_classes,

)

model = dict(
    type=ImageClassifier,
    data_preprocessor=data_preprocessor,
    backbone=dict(
        type=MobileNetv2, widen_factor=widen_factor, out_indices=(2,), rep=False
    ),
    neck=dict(type=GlobalAveragePooling),
    head=dict(
        type=LinearClsHead,
        in_channels=8,
        num_classes=num_classes,
        loss=dict(
            type=CrossEntropyLoss,
            loss_weight=1.0,
        ),
    ),
)

train_cfg = dict(by_epoch=True, max_epochs=epochs)

param_scheduler = [
    dict(type=LinearLR, begin=0, end=30, start_factor=0.001, by_epoch=False),  # warm-up
    dict(
        type=MultiStepLR,
        begin=1,
        end=100,
        milestones=[50, 70, 90],
        gamma=0.3,
        by_epoch=True,
    ),
]

```

LynnL4, 2025-01-16:
Hi, have you exported the vela model? Grove Vision AI needs to convert the model to vela. Try setting the format to Vela when exporting, and you will get a model named xx_vela.tflite.

rn776, 2025-01-16:
Yes I exported to vela format with the command
```
python3 tools/export.py     configs/pipeineer/mobnetv2_custom.py     work_dirs/mobnetv2_custom/epoch_50.pth     --cfg-options     data_root=/home/pipe_dataset     num_classes=2     epochs=50     --imgsz 192 192     --format vela     --image_path /home/pipe_dataset/valid

```

There are some warnings and it says that Export of vela model succeeded. I flashed the resulting epoch_50_int8_vela.tflite using the web toolkit and get the 'Invoke Failed. please check device connection'.
I get this output when running the above command. :

```
ModelAssistant/mmengine/optim/optimizer/zero_optimizer.py:11: DeprecationWarning: `TorchScript` support for functional optimizers is deprecated and will be removed in a future PyTorch release. Consider using the `torch.compile` optimizer instead.
  from torch.distributed.optim import \
01/16 08:33:52 - mmengine - INFO - 
------------------------------------------------------------
System environment:
    sys.platform: linux
    Python: 3.11.11 | packaged by conda-forge | (main, Dec  5 2024, 14:17:24) [GCC 13.3.0]
    CUDA available: True
    MUSA available: False
    numpy_random_seed: 414715920
    GPU 0: NVIDIA GeForce RTX 2070 with Max-Q Design
    CUDA_HOME: /usr
    NVCC: Cuda compilation tools, release 10.1, V10.1.24
    GCC: gcc (Ubuntu 9.4.0-1ubuntu1~20.04.2) 9.4.0
    PyTorch: 2.5.1+cu124
    PyTorch compiling details: PyTorch built with:
  - GCC 9.3
  - C++ Version: 201703
  - Intel(R) oneAPI Math Kernel Library Version 2024.2-Product Build 20240605 for Intel(R) 64 architecture applications
  - Intel(R) MKL-DNN v3.5.3 (Git Hash 66f0cb9eb66affd2da3bf5f8d897376f04aae6af)
  - OpenMP 201511 (a.k.a. OpenMP 4.5)
  - LAPACK is enabled (usually provided by MKL)
  - NNPACK is enabled
  - CPU capability usage: AVX2
  - CUDA Runtime 12.4
  - NVCC architecture flags: -gencode;arch=compute_50,code=sm_50;-gencode;arch=compute_60,code=sm_60;-gencode;arch=compute_70,code=sm_70;-gencode;arch=compute_75,code=sm_75;-gencode;arch=compute_80,code=sm_80;-gencode;arch=compute_86,code=sm_86;-gencode;arch=compute_90,code=sm_90
  - CuDNN 90.1
  - Magma 2.6.1
  - Build settings: BLAS_INFO=mkl, BUILD_TYPE=Release, CUDA_VERSION=12.4, CUDNN_VERSION=9.1.0, CXX_COMPILER=/opt/rh/devtoolset-9/root/usr/bin/c++, CXX_FLAGS= -D_GLIBCXX_USE_CXX11_ABI=0 -fabi-version=11 -fvisibility-inlines-hidden -DUSE_PTHREADPOOL -DNDEBUG -DUSE_KINETO -DLIBKINETO_NOROCTRACER -DLIBKINETO_NOXPUPTI=ON -DUSE_FBGEMM -DUSE_PYTORCH_QNNPACK -DUSE_XNNPACK -DSYMBOLICATE_MOBILE_DEBUG_HANDLE -O2 -fPIC -Wall -Wextra -Werror=return-type -Werror=non-virtual-dtor -Werror=bool-operation -Wnarrowing -Wno-missing-field-initializers -Wno-type-limits -Wno-array-bounds -Wno-unknown-pragmas -Wno-unused-parameter -Wno-strict-overflow -Wno-strict-aliasing -Wno-stringop-overflow -Wsuggest-override -Wno-psabi -Wno-error=old-style-cast -Wno-missing-braces -fdiagnostics-color=always -faligned-new -Wno-unused-but-set-variable -Wno-maybe-uninitialized -fno-math-errno -fno-trapping-math -Werror=format -Wno-stringop-overflow, LAPACK_INFO=mkl, PERF_WITH_AVX=1, PERF_WITH_AVX2=1, TORCH_VERSION=2.5.1, USE_CUDA=ON, USE_CUDNN=ON, USE_CUSPARSELT=1, USE_EXCEPTION_PTR=1, USE_GFLAGS=OFF, USE_GLOG=OFF, USE_GLOO=ON, USE_MKL=ON, USE_MKLDNN=ON, USE_MPI=OFF, USE_NCCL=1, USE_NNPACK=ON, USE_OPENMP=ON, USE_ROCM=OFF, USE_ROCM_KERNEL_ASSERT=OFF, 

    TorchVision: 0.20.1+cu124
    OpenCV: 4.10.0
    MMEngine: 0.10.4

Runtime environment:
    cudnn_benchmark: True
    mp_cfg: {'mp_start_method': 'fork', 'opencv_num_threads': 0}
    dist_cfg: {'backend': 'nccl'}
    seed: 414715920
    deterministic: False
    Distributed launcher: none
    Distributed training: False
    GPU number: 1
------------------------------------------------------------

2025-01-16 08:33:52.582754: E external/local_xla/xla/stream_executor/cuda/cuda_fft.cc:477] Unable to register cuFFT factory: Attempting to register factory for plugin cuFFT when one has already been registered
WARNING: All log messages before absl::InitializeLog() is called are written to STDERR
E0000 00:00:1737016432.597606   31807 cuda_dnn.cc:8310] Unable to register cuDNN factory: Attempting to register factory for plugin cuDNN when one has already been registered
E0000 00:00:1737016432.602591   31807 cuda_blas.cc:1418] Unable to register cuBLAS factory: Attempting to register factory for plugin cuBLAS when one has already been registered
2025-01-16 08:33:52.618368: I tensorflow/core/platform/cpu_feature_guard.cc:210] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
01/16 08:33:54 - mmengine - INFO - Distributed training is not used, all SyncBatchNorm (SyncBN) layers in the model will be automatically reverted to BatchNormXd layers if they are used.
=============================================================================================================================
Layer (type:depth-idx)                             Input Shape               Output Shape              Param #
=============================================================================================================================
ImageClassifier                                    [3, 3, 192, 192]          [3, 2]                    --
[... cut ...]
   Op has tensors with missing quantization parameters: model_5/tf.math.reduce_mean/Mean, arith.constant21_reshape, PartitionedCall:0

Network summary for epoch_50_int8
Accelerator configuration                Ethos_U55_64
System configuration                       My_Sys_Cfg
Memory mode                        My_Mem_Mode_Parent
Accelerator clock                                 400 MHz

CPU operators = 27 (100.0%)
NPU operators = 0 (0.0%)

Neural network macs                                 0 MACs/batch

Export of vela model succeeded

```
It seems there was some error while exporting the model?

mjq2020, 2025-01-16:
Hello, according to the output log, it seems that the verification image you gave was not detected by the program. The main reason is that during the verification process, the program only obtains images with the jpg suffix. This is our problem and we will fix this bug later. You can also modify the code [here](https://github.com/Seeed-Studio/ModelAssistant/blob/89716d373fa84c5860db90d0a38519b04812f8c4/tools/export.py#L364) according to your image suffix，and change jpg to the suffix of your image format.

rn776, 2025-01-16:
Thank you for the response. 
I had previously trained using png image. I retrained the models with jpeg images and deleted the old calibration_image_sample_data_20x128x128x3_float32.npy file before exporting the model. It now invokes in the Sensecraft web toolkit.