Seeed-Studio/ModelAssistant #278: the output of person detection model based yolov5 not same with PC version

jianjunchang, 2025-01-20:
hi, i have deploy the model (same as person_Detection_Swift-YOLO_Nano_192.ipynb) on a esp32s3 board, an it can detect a person very well, but the box (x, y, w, h) position not very good, it always on the left of the real target position, especially the target close to the camera. for testing, i have deploy the same tflite model on PC, all the preprocess, inference and postprocess are same as i done on the esp32s3, and the result is very good, box (x, y, w, h) position is perfect (not on the left of the real target position again), i was confused, and need help. 
1. why did i got the different result with the same tflite model on PC and esp32s3?
2. how can i fix my inference code logic for geting right box position?
3. i found the speed of the same tflite model are differently on different environment, (PC python, about 0.05s per image) > (esp32s3 idf, about 1.2s per image) >  (PC c++, about 1.9s per image)，the result is also confuse me.
my english is not good, i hope use chinese for adding some more descripton(include some screenshot) of my question, thanks for your attention.

baorepo, 2025-01-20:
@jianjunchang   I don't know where to start answering your questions, so I suggest you post your code and the relevant results. Let's do this case by case.

jianjunchang, 2025-01-22:
> [@jianjunchang](https://github.com/jianjunchang) I don't know where to start answering your questions, so I suggest you post your code and the relevant results. Let's do this case by case.

hi, i have solved the problem, just delete all esp_timer_get_time() code, which i used for testing the time consuming when runing the ai model, may be it's not reasonable for my problem, and i'm sure it's work much better than before. do you know why the esp_timer_get_time function will cause the problem? 
i have done a testing, run my model with a single picture other than capture from a camera, it will out several different results(before delete the esp_timer_get_time() code), this is a very strange  phenomenon. now, the outs seemly the same (just a bit differents) when i do the testing again, and this is a normal phenomenon.

LynnL4, 2025-01-22:
Hi, Which example program did you base your modifications on? As far as we know, there is a version of ESP's TFLite-Micro that can cause similar issues. Please refer to [[this GitHub issue](https://github.com/espressif/esp-nn/issues/6)](https://github.com/espressif/esp-nn/issues/6) for more details.

closed 2025-02-13 by github-actions[bot]