Seeed-Studio/ModelAssistant #279: tflite-runtime is incompatible with Python 3.12

aircable, 2025-01-21:
On Ubuntu Linux 24.04.1 LTS going through the FOMO Model Training Tutorial I cannot verify the models.
Conda is activated: `conda create --name sscma python=3.12 -y`

The error is, with all my tries to resolve the issue:

WARNING: Package(s) not found: tflite_runtime
ERROR: Could not find a version that satisfies the requirement tflite_runtime (from versions: none)
ERROR: No matching distribution found for tflite_runtime
Please install tflite_runtime manually

`(sscma) ModelAssistant$ pip3 install tflite_runtime`
ERROR: Could not find a version that satisfies the requirement tflite_runtime (from versions: none)
ERROR: No matching distribution found for tflite_runtime

`(sscma) ModelAssistant$ pip3 install --extra-index-url https://google-coral.github.io/py-repo/ tflite_runtime`
Looking in indexes: https://pypi.org/simple, https://google-coral.github.io/py-repo/
ERROR: Could not find a version that satisfies the requirement tflite_runtime (from versions: none)
ERROR: No matching distribution found for tflite_runtime

`(sscma) ModelAssistant$ sudo apt install python3-tflite-runtime`
[sudo] password for juergen: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package python3-tflite-runtime

The only possible solution would to switch to the new LiteRT. On Python 3.12 only the nightly build installs:
`(sscma) ModelAssistant$ pip3 install ai-edge-litert-nightly`

It would require to change the code in tools/export.py to use the new Interpreter class.
There are still too many dependencies on tflite_runtime. I was not able to resolve the issue.
Would you guys take a look at that?

As a side note and an additional edit for your Installation page, tinynn requires a special version for the quantization-aware training phase:
`(sscma) ModelAssistant$ pip3 install torch --index-url https://download.pytorch.org/whl/cpu`
`(sscma) ModelAssistant$ pip3 install psutil`
`(sscma) ModelAssistant$ pip3 install git+https://github.com/alibaba/TinyNeuralNetwork.git`

LynnL4, 2025-01-22:
Thank you for your feedback. I have successfully reproduced your issue on Ubuntu 24.04 LTS and am currently working on a fix.

LynnL4, 2025-02-05:
Hi, I already fixed it. please pull and try again, Thanks.

aircable, 2025-02-07:
Thank you, so far so good. I had to change this line:
        #OPTIMIZERS.register_module(name='Adafactor', module=Adafactor)
because the Adafactor is already registered in optimizer at torch.optim

closed 2025-02-12 by LynnL4