Seeed-Studio/ModelAssistant #282: MobilenetV3 Grove AI vision V2 board

rn776, 2025-02-17:
Hello, I have a classification model with MobilenetV3 as the backbone. The trained model was exported to 'vela' format as suggested. However, the flashed board fails to invoke. The classifier model with MobilenetV2 invokes without issue. Is there any way to run the MobilenetV3 models on the board?

baorepo, 2025-02-19:
@rn776 There is not much difference between v3 and v2. Do you have a visual model? How about a comparison?

rn776, 2025-02-21:
[image]
[image]

Here are the two models. The mobilenetv3 flashes into the board but it says 'invoke failed'.

closed 2025-03-18 by github-actions[bot]