Seeed-Studio/ModelAssistant #296: Error when install python package and Pytorch

imnotwannafire, 2025-08-07:
Hi, 
When I run command `!. ./scripts/setup_colab.sh` on current google colab. Some packages cannot be installed with following log
`ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
ipython 7.34.0 requires jedi>=0.16, which is not installed.
google-colab 1.0.0 requires requests==2.32.3, but you have requests 2.28.2 which is incompatible.
pymc 5.25.1 requires rich>=13.7.1, but you have rich 13.4.2 which is incompatible.
dataproc-spark-connect 0.8.3 requires tqdm>=4.67, but you have tqdm 4.65.2 which is incompatible.
opencv-contrib-python 4.12.0.88 requires numpy<2.3.0,>=2; python_version >= "3.9", but you have numpy 1.26.4 which is incompatible.
pytensor 2.31.7 requires filelock>=3.15, but you have filelock 3.14.0 which is incompatible.
sphinx 8.2.3 requires requests>=2.30.0, but you have requests 2.28.2 which is incompatible.
yfinance 0.2.65 requires requests>=2.31, but you have requests 2.28.2 which is incompatible.
datasets 4.0.0 requires requests>=2.32.2, but you have requests 2.28.2 which is incompatible.
datasets 4.0.0 requires tqdm>=4.66.3, but you have tqdm 4.65.2 which is incompatible.
thinc 8.3.6 requires numpy<3.0.0,>=2.0.0, but you have numpy 1.26.4 which is incompatible.`

When running train command, it automatically disable Pytorch.
`Disabling PyTorch because PyTorch >= 2.1 is required but found 2.0.0`

[image]

imnotwannafire, 2025-08-11:
> ```
> !git clone https://github.com/Seeed-Studio/ModelAssistant.git -b 2.0.0  #clone the repo
> %cd ModelAssistant
> !. ./scripts/setup_colab.sh
> ```
> 
> When you download the source code, please make sure to use the 2.0.0 branch. [@imnotwannafire](https://github.com/imnotwannafire)

Yes, I have cloned branch 2.0.0
I followed the guide notebook of seedstudio in [here](https://github.com/Seeed-Studio/sscma-model-zoo/blob/main/notebooks/en/person_Detection_Swift-YOLO_192.ipynb)

baorepo, 2025-08-11:
[image] @imnotwannafire  running the train step, works fine

imnotwannafire, 2025-08-11:
> t a try and there was no problem.

Hi, 
plz scroll up to above logs and you will find the error logs. 
It did not appear in the end of log

imnotwannafire, 2025-08-11:
> [@imnotwannafire](https://github.com/imnotwannafire) Sorry, I closed this notebook. You can actually reproduce it very easily. Just run it in colab.

I have reproduce the error on colab. please check the log

[image]

[image]

baorepo, 2025-08-12:
https://colab.research.google.com/gist/baorepo/afc93acb12ed6068ae366517c68a3ac7/person_detection_swift-yolo_192.ipynb @imnotwannafire  I just ran it again and noticed the issue you mentioned. The essence of this problem is a pip dependency issue, specifically related to the numpy 2.0 version. Clearly, our software does not support numpy 2.0, so this error can be ignored.

imnotwannafire, 2025-08-12:
thank you for your reply, which means the training process is not affected by this error? And the model still work fine?

baorepo, 2025-08-13:
Yes, you can actually verify it. @imnotwannafire

baorepo, 2025-08-13:
For resolutions as small as 192x192, the models available on the market are quite similar. The demand is very niche and thus fails to catch the attention of researchers. @imnotwannafire

imnotwannafire, 2025-08-14:
Thank you baorepo! I will close the issue

> For resolutions as small as 192x192, the models available on the market are quite similar. The demand is very niche and thus fails to catch the attention of researchers. [@imnotwannafire](https://github.com/imnotwannafire)